xwiimote/xwiimote #66: 3rd Party Wii U Pro not connecting

YayIguess, 2015-12-22:
(Fedora 23, Dell Latitude E6330, buez 5, using built-in bluetooth, I just compiled xwiimote in last week or so, I have one of these http://imgur.com/wtRNKQe)

Blueman can't find my Wii U pro controller, yet it finds and connects to my regular wii remote just fine.

`hcitool  inq --iac=liac` returns:  
 `40:F4:07:CD:38:9E	clock offset: 0x1f86	class: 0x002504`

`sudo hcitool info 40:F4:07:CD:38:9E`  returns:  
`BD Address:  40:F4:07:CD:38:9E
	OUI Company: Nintendo Co., Ltd. (40-F4-07)
	Device Name: Nintendo RVL-CNT-01-UC
	LMP Version: 3.0 (0x5) LMP Subversion: 0xc
	Manufacturer: not assigned (771)
	Features: 0xbc 0x02 0x04 0x38 0x08 0x00 0x00 0x00
		<encryption> <slot offset> <timing accuracy> <role switch> 
		<sniff mode> <RSSI> <power control> <enhanced iscan> 
		<interlaced iscan> <interlaced pscan> <AFH cap. slave> `

`connect 40:F4:07:CD:38:9E` inside `bluetoothctl` returns:
`Device 40:F4:07:CD:38:9E not available`

Any idea how to get it working? Or will I just need to buy a different one?

minexew, 2016-08-17:
This seems to be pretty much the same issue as described in #43.

firestar115, 2016-11-06:
I have this same issue with a Hyperkin ProCube. `hcitool inq -iac=liac` returns:

    00:3C:10:80:25:AA    clock offset: 0x3e9d    class: 0x002504

`sudo hcitool cc 00:3C:10:80:25:AA` turns the controller off after pressing the sync button with my phone paired, but connects the controller for about a second without my phone paired. I have no idea whether my computer disconnects it or whether it disconnects itself.

`[sudo] bluetoothctl`, `connect 00:3C:10:80:25:AA` reports that the device is unavailable.

#43 helped me by giving me these commands, but gave no information on how to fix the problem. I would like for this problem to be fixed so I can play Melee with a controller and not a keyboard.

firestar115, 2016-11-06:
Here is the solution:

1. Have two terminals open.
2. In the first terminal, execute `sudo bluetoothctl`
3. In the second, type in `sudo sudo hcitool cc 40:F4:07:CD:38:9E` but don't press enter yet.
4. In `bluetoothctl`, type in `pair 40:F4:07:CD:38:9E` but also don't press enter yet.
5. Press they SYNC button on your controller.
6. Press enter in the `hcitool` terminal.
7. Press enter after half a second in the `bluetoothctl` terminal. Keep pressing Up and Enter until the device pairs.
8. You should now be able to `connect` in bluetoothctl` without a problem.

But what do we do next to keep the controller on?

firestar115, 2016-11-21:
So, once the controller is connected, how do we map it to keyboard/mouse or an XBOX controller input?

`xwiishow list` after connecting to the controller outputs no controllers.

TsundereBug, 2017-06-16:
Half a year and it still connects. Fedora 25. Going to see if I can get it mapped to a gamepad without xwiimote first.

TsundereBug, 2017-06-16:
Connecting to the controller and running `xwiishow list` outputs that there are no devices. The device does show up in Control Center as `Nintendo RVL-CNT-01-UC` however.

TsundereBug, 2017-06-17:
On my new computer Linux installs keep breaking on reboot. I might have an answer this afternoon. According to a Reddit post, we don't need xwiimote, but as far as I know the computer doesn't take any type of input from it or send any data to keep it alive.

TsundereBug, 2017-06-17:
Here are my results (@dvdhrm I don't think xwiimote installs properly on newer versions of Linux)

`RVL-CNT-01` Connects flawlessly, but I vaguely remember this being a stock thing. `xwiishow` and all commands work properly for it however.
`RVL-CNT-01-TR` Asks for a password (which if I remember correctly from the spec is the mac address in bytes backwards) so I cannot get it to connect.
`RVL-CNT-01-UC` Connects but does not stay connected. `xwiishow` does not list it during the connection.

So it seems like xwiimote does not _do_ anything. And I'm not a C/C++ programmer so I wouldn't know where to start to fix it. If anyone has a stock WiiU Pro Controller around, please test with latest version of xwiimote and updated versions of everything else. This might not be a 3rd party problem anymore.

ildar, 2020-01-15:
I got similar problem with a WiiMote clone (based on RDA 5875Y chip).
1st I couldn't find it during BT discovery. At last, using `hcitool inq --iac=liac` I got its BT address. Then:
```
#  hcitool info 00:17:AB:39:CB:52
Requesting information ...
        BD Address:  00:17:AB:39:CB:52
        OUI Company: Nintendo Co., Ltd. (00-17-AB)
        Device Name: Nintendo RVL-CNT-01
        LMP Version: 3.0 (0x5) LMP Subversion: 0xc
        Manufacturer: IACA electronique (771)
        Features: 0xbc 0x02 0x04 0x38 0x08 0x00 0x00 0x00
                <encryption> <slot offset> <timing accuracy> <role switch>
                <sniff mode> <RSSI> <power control> <enhanced iscan>
                <interlaced iscan> <interlaced pscan> <AFH cap. slave>
```

Fine. 2nd, `hcitool cc 00:17:AB:39:CB:52` allowed me to see it in `bluetoothctl` pair/connect.
But failure is in that while `bluetoothctl` reports successful connect, the gamepad remains blinking. 
bluez also complains:
> bluetoothd[188208]: 00:17:AB:39:CB:52: error updating services: Input/output error (5)

Anyone propose my next step, please?

waltercool, 2020-05-31:
Same problem as above. Sorry for repeating the message, that's my input

# hcitool info 00:17:AB:39:A6:D6
Requesting information ...
        BD Address:  00:17:AB:39:A6:D6
        OUI Company: Nintendo Co., Ltd. (00-17-AB)
        Device Name: Nintendo RVL-CNT-01
        LMP Version: 3.0 (0x5) LMP Subversion: 0xc
        Manufacturer: IACA electronique (771)
        Features: 0xbc 0x02 0x04 0x38 0x08 0x00 0x00 0x00
                <encryption> <slot offset> <timing accuracy> <role switch> 
                <sniff mode> <RSSI> <power control> <enhanced iscan> 
                <interlaced iscan> <interlaced pscan> <AFH cap. slave>

Wiimote works great if bypassing Dolphin Emulator bluetooth adapter, but seems like it can't handle it by Bluez.

Lazorne, 2020-08-17:
Same as above on both of mine:

```
BD Address:  00:17:AB:39:C5:44
	OUI Company: Nintendo Co., Ltd. (00-17-AB)
	Device Name: Nintendo RVL-CNT-01
	LMP Version: 3.0 (0x5) LMP Subversion: 0xc
	Manufacturer: IACA electronique (771)
	Features: 0xbc 0x02 0x04 0x38 0x08 0x00 0x00 0x00
		<encryption> <slot offset> <timing accuracy> <role switch> 
		<sniff mode> <RSSI> <power control> <enhanced iscan> 
		<interlaced iscan> <interlaced pscan> <AFH cap. slave> 

BD Address:  00:17:AB:39:97:61
	OUI Company: Nintendo Co., Ltd. (00-17-AB)
	Device Name: Nintendo RVL-CNT-01
	LMP Version: 3.0 (0x5) LMP Subversion: 0xc
	Manufacturer: IACA electronique (771)
	Features: 0xbc 0x02 0x04 0x38 0x08 0x00 0x00 0x00
		<encryption> <slot offset> <timing accuracy> <role switch> 
		<sniff mode> <RSSI> <power control> <enhanced iscan> 
		<interlaced iscan> <interlaced pscan> <AFH cap. slav

```

mrfixit2001, 2021-02-08:
Just to add some detail... I spent a good amount of time trying different things... In case this information somehow leads another developer to success.

I have a gamepad that registers as a Nintendo RVL-CNT-01-UC with MAC 40:F4:07:CD:4A:0D, and this little trick at least got it "connecting" - but then the lights blink for another few seconds and it disconnects after that (same behavior others have noticed).
https://github.com/dvdhrm/xwiimote/issues/66#issuecomment-258695985

So I edited the source for Bluez, tried the latest version as well as a few older ones, and even made some changes to the kernel driver itself. Overall - no luck. I hacked bluetoothd to track exactly what it's doing... bottom line is that the device isn't getting added to the kernel whitelist, and telling the kernel to ignore the whitelist just makes no devices connect.

The Bluez driver contains a comment that immediately after connecting the gamepad needs to connect to the input service, otherwise it will disconnect in a few seconds (sounds familiar). Perhaps that's not happening - but I don't really know how to confirm this. I can at least confirm that even after bluez briefly does connect the gamepad, the kernel doesn't pick up a new HID added - so I expect this is not happening.

I finally noticed something that's very different about this controller... when I plugged in the USB cable, the kernel found something! That's not normal behavior for a Wii U controller from what I can tell. My other Wii U controller just charges over USB and doesn't have any IO at all over the interface. Upon further inspection, the Vendor and Product ID on this gamepad (idVendor=0471, idProduct=0666) register to the kernel as a "Philips Hantek DDS-3005 Arbitrary Wave Generator". Hillarious, right?

So, I actually edited the wiimote kernel driver and forced this product and vendor ID to be recognized as a wiimote instead. The kernel finds it, installs the wiimote, registers the HID, but then the exact same behavior happens... lights flash for a few seconds, then turn off and the kernel removes it. 

Next, I added code to the wiimote bluez plugin to mimics the behavior of the sixaxis plugin - whitelist and bind the device when it's recognized via USB. The code ran, but using USB it was unable to pull the physical address off the device and the whitelisted address was all zeros.

I should also point out my other Wii U clone from a different mfg connects just fine - altho it asks for a pin when pairing and doesn't accept anything I tried - so it's not able to auto re-connect later - but it connects and works fine. It doesn't have any active IO over usb.

So at this point, I'm throwing in the towel and just making the determination that this thing has completely screwed up firmware. The only thing I can think *might* work would be to patch bluez to not care if it can be added to the kernel whitelist AND patch the kernel driver to not care if a device is in the whitelist or not. Both patches might possibly work, or improve the situation, but that would be a terrible idea to leave those patches in place outside of just testing. So yeah - this controller is now officially a toy for my children to destroy - or to think they are playing the game that I'm playing :D

s5bug, 2021-02-09:
Perhaps it may be worth it to raise this up to BlueZ? Or at least try and write some sort of tool that uses the things we know now, and see if it can be done, and xwiimote is just very out of date?

I don't own a Wii U, and I'm pretty sure my ProCube is basically dead (I use a real GameCube controller now, you can get great adapters for them for like $10 [although a really good controller will run you >$70]), but if it's not and I get through my other projects I'm definitely willing to try this again.

By any chance @mrfixit2001, do you know how to develop for bluetooth on NixOS? Or should I make space for some other OS? What should a dev flow for a tool like this be?