xwiimote/xwiimote #70: XWiimote website reports compatibility with drums and guitar extensions, but the kernel does not actually support them

meden, 2016-04-09:
Hello,

the title pretty says it all.

While trying to make my guitar work in Linux, I explored the [git log of `hid-wiimote-core.c`](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/log/drivers/hid/hid-wiimote-core.c) finding that the commits related to guitar and drum support [have been reverted](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/hid/hid-wiimote-core.c?id=b04c99e3b845892d754ee8052d6324c39c4040de).

That's a pity... So, any plan to re-add that functionality, maybe [implementing ABS_MAX2/CNT2](http://www.gossamer-threads.com/lists/linux/kernel/1793951) (whatever that ever means :) )?

Thanks for your work.

P.S.: if you can give any pointer or simple explanation of what caused the commit revert, and the solution you proposed, it would be really appreciated.

nadenislamarre, 2016-04-09:
The commit was reverted because it broke some other devices (bluetooth
keyboards as far as i remember).
My understanding was that the drums have 5 pads, while there was a kind of
restriction to 4 in the linux kernel due to some algorithms coded for 4 (so
not just a variable to change).
David Herrmann can probably give more information while he was in charge to
review my patch.
I guess however that my patch is still working if you recompile your module.
I've no idea if that would be accepted if we remove 1 pad (=disable one)

Nicolas

andreapergola, 2017-04-30:
Would be nice to see this added, any news or hopes for this? :)

Darkhogg, 2018-01-23:
I understand this is an old issue, but I'd like to bump – Any hopes of seeing the guitar at least??

meden, 2018-01-27:
Maybe could be of some interest knowing that the question has been raised up again just after I filed this issue: https://lkml.org/lkml/2016/9/27/553

castlec, 2018-10-30:
It looks like everything was changed in the kernel and in xwiishow to use drums again. I'm on ubuntu and have used their source packages to rebuild against the current master here. xwiishow is now giving me the ability to toggle drums on. I'm not seeing any input from the drums though. 

I'm not sure if I have a hardware problem or if my kernel is not right. I'm on 4.19.0-041900-generic from ubuntu 18.10.

Advice appreciated. I imagine this issue can be closed since, as far as I can tell, it should be functional now.

dvdhrm, 2018-10-30:
There was drums+guitar support back in the days in the upstream kernel, and xwiishow supported it. It got reverted since it didn't work as expected. However, a different patchset was merged just few months ago. So new kernels support drums+guitars, but they report different IDs now. Hence, xwiishow probably does not show them correctly, but it should be trivial to adjust it.

castlec, 2018-10-30:
I imagine you're right but I haven't been successful. 

What I've observed while poking around is that the extensions aren't receiving events at all. The individual devices are identified and the interfaces are opened. The wiimote itself generates events but even with a nunchuck, I'm not receiving events. 

I'm not sure what to do at this point. Advice appreciated.

mishugashu, 2019-09-21:
I tried today and got the guitar (mostly) working. Anyone try recently? I'm on kernel `5.3.0-arch1-1-ARCH`. I can get red and green buttons going, the strum working, the whammy stick, the start+select buttons, and the analogue knob. But... orange, blue, and yellow buttons don't seem to work. I haven't used this guitar in a number of years, though, so it might be broken?