xwiimote/xwiimote #74: balanceboard: cannot read led state

Esylias, 2017-04-05:
i'm connected to wiiboard and "xwiishow 1" find a balanceboard device. no scale, but an error "cannot read led state" => [Screenshot](http://fs5.directupload.net/images/170405/g25vi6r6.jpg)

if i connect to board by bluetoothctl, the out from "hcidump -i hci0" get filled with

> ACL data: handle 12 flags 0x02 dlen 8
> L2CAP(d): cid 0x0041 len 4 [psm 19]
> HIDP: Data: Input report

is it possible to make this output human?

iomihai, 2017-04-05:
As far as I can tell the message "cannot read led state" is normal for the balance board.
Can you not see the sum of the weight sensors changing their value in the Balance Board section (center screen)?
It's the sum of the sensors in each corner and the value is the weight is kilos divided by 100 ± some calibration value. 
For me the value shown is approx 1.5kg less then the real weight.

Shameless plug :) , I built a Raspberry Pi interface using xwiimote for the balance board that saves the weight on a Google spreadsheet. You might try it https://github.com/iomihai/rpiscale .

Esylias, 2017-04-05:
@iomihai 
the weight sensors does not changing their values, pls have a look at my [Screenshot](http://fs5.directupload.net/images/170405/g25vi6r6.jpg)

iomihai, 2017-04-05:
Did you trust the balance board? https://wiki.archlinux.org/index.php/bluetooth#Bluetoothctl

Esylias, 2017-04-05:
Yes its trusted and i got an successfully connection to the board