xwiimote/xwiimote #75: Motion Plus Slow / Fast Mode

Franck7L, 2017-05-22:
Hello ! 

I'm trying to use xwiimote to get orientation angles from gyro of RVL-CNT-01-TR wiimotes.
Now that I'm a little more documented on wiimote hardwares and protocols, i've seen MotionPlus got  two "speed" modes (slow or fast). If I understand it correctly, the mode is auto-switched by hardware according to velocity of user movements and it modifies the scale of the rate output.
I've seen some old libraries reading the speed mode info on the fly and providing normalisation to ease values integration, but I didn't find references to these modes in API or core code of xwiimote, so I don't know if and how it's handled.

Do you have some knowledge on it ?

dvdhrm, 2017-05-29:
Your kernel normalizes the data already. See here:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hid/hid-wiimote-modules.c#n2034

Franck7L, 2017-05-29:
Thank You David, so I understand I don't have to care about.

dvdhrm, 2017-05-30:
Yes, if the code is correct, the data should already be returned in normalized form.

closed 2017-05-30 by dvdhrm