xwiimote/xwiimote #83: MotionPlus data are inconsistent between a discrete Extension and a Remote Plus

epigramx, 2018-01-16:
It appears that a discrete Extension that works fine on Dolphin with Bluetooth Passthrough and a new Remote Plus remote (carefully tested in Skyward Sword), do not appear to return MotionPlus data in the same way. One can easily see for example that holding it in an "aiming" position the xwiishow shows a remote plus' indicator going to the left when turning left, but a vanilla remote with the discrete extension shows the X going to the right when turning left!

epigramx, 2018-01-16:
I have a suspicion the true reason is that calibration data from MotionPlus is not fully reverse engineered. I've never seen a HID library for that device that handles calibration data from MP, or not fully at least.

jordan-woyak, 2019-11-17:
As it has been mentioned, this is caused by not interpreting the calibration data.
Various Wii remotes have gyro axes "reversed" with matching "reversed" calibration data.

FYI, the calibration data has recently been fully reverse engineered by the dolphin-emu team.

I've documented the calibration data format here: http://wiibrew.org/wiki/Wiimote/Extension_Controllers/Wii_Motion_Plus

v1993, 2022-01-27:
This should be fixed in kernel driver rather than userspace library, which is what does hardware-level interactions with wiimote: linux-input@vger.kernel.org.