xwiimote/xwiimote #84: balance board not sending any EVENT_BALANCE_BOARD events

blendmaster, 2018-04-24:
This is half a possible hardware support question and half a xwiimote question.

I hooked up my old balance board to xwiimote recently and was able to get measurements out of it up until yesterday, when, with identical client code, I haven't been able to get a single EVENT_BALANCE_BOARD to fire. Pairing with the balance board still works find and pressing the power button still sends a button event, but the stream of measurement events is gone.

I thought it was a hardware/firmware issue of some sort, so I took the batteries out of the board for 24 hours, but to no avail. I've also tried unloading xwiimote-hid kernel module, unpairing and repairing, even  trying some of the non-xwiimote bluez-based readers like https://github.com/aelveborn/Wii-Scale. They also seem in the same situation as xwiimote in that they can pair with the board, but aren't getting any weight measurements out. All of these tests are from the same bluetooth host computer, so something could be up with that. I'll try to test on another machine when I get the chance.

So more of a hardware support question: does the balance board require any sort of handshake signaling to start sending weight events that I'm missing? Similarly is there some firmware/software reset that I can do on the device to make it send events again? I _think_ that's the problem here, but here's more environment details just in case:

    Linux kami 4.13.0-38-generic #43~16.04.1-Ubuntu SMP Wed Mar 14 17:48:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
    xwiimote built from f2be57e24fc24652308840cec2ed702b9d1138df

I'm using xwiishow to debug, but I also have this short python script adapted from the test script I was using before it broke:

```python
#!/usr/bin/env python2
"""
print the total force on the wii balance board, in earth kg,
smoothed by a moving average.
"""

import errno
from time import sleep
from select import poll, POLLIN
from inspect import getmembers
from pprint import pprint
import xwiimote

try:
    mon = xwiimote.monitor(True, True)
    ent = mon.poll()
    firstwiimote = ent
    while ent is not None:
        print "Found device: " + ent
        ent = mon.poll()
except SystemError as e:
    print "ooops, cannot create monitor (", e, ")"

# continue only if there is a wiimote
if firstwiimote is None:
    print "No wiimote to read"
    exit(1)

# create a new iface
try:
    dev = xwiimote.iface(firstwiimote)
except IOError as e:
    print "ooops,", e
    exit(1)

# display some information and open the iface
try:
    print "syspath:" + dev.get_syspath()
    fd = dev.get_fd()
    print "fd:", fd
    print "opened mask:", dev.opened()
    dev.open(dev.available() | xwiimote.IFACE_WRITABLE)
    print "opened mask:", dev.opened()
except SystemError as e:
    print "ooops", e
    exit(1)

# read some values
p = poll()
p.register(fd, POLLIN)
evt = xwiimote.event()

# jank circular buffer
buf = [0] * 50
pos = 0
while True:
    p.poll()
    try:
        dev.dispatch(evt)
        if evt.type == xwiimote.EVENT_BALANCE_BOARD:
            a = evt.get_abs(0)[0]
            b = evt.get_abs(1)[0]
            c = evt.get_abs(2)[0]
            d = evt.get_abs(3)[0]
            tot = (a + b + c + d) / 100.0 # decigrams
            buf[pos] = tot
            pos = (pos + 1) % len(buf)
            import sys
            sys.stdout.write("\rWeight: %.2fkg" % (sum(buf) / len(buf)))
        else:
            print("unknown event: %d" % evt.type)
    except IOError as e:
        if e.errno != errno.EAGAIN:
            print "Bad"

exit(0)
```

blendmaster, 2018-04-24:
Ah, after doing more searching, I tried out https://github.com/rpavlik/wiiuse and it _is_ able to read out balance board events. So it's possible there is some sort of handshake that xwiimote is missing. I'll keep digging.

blendmaster, 2018-04-24:
Welp, after connecting with wiiuse a couple times, now xwiishow and that xwiimote python script are reporting weight values again. Mysterious. I don't know if the handshake somehow carries over, or there was just some temporary lapse in the hardware. I suppose if you the reader come across this and are having the same issue, try out wiiuse and see if it "heals" the balance board, then come back to xwiimote so you don't have to push the red sync button every time to connect.

vladigor, 2018-05-28:
Hi @blendmaster .  It sounds like I've got the same problem - wiishow doesn't report anything from the balance boards weight sensors.   
I've tried installing wiiuse as you suggest but am having difficulties using it.  When I run wiiuseexample I get returned "Found 0 bluetooth device(s). No wiimotes found."  The balance board is paired and connected.  Was there a trick to getting wiiuse to work?

vladigor, 2018-05-30:
Well, my board is working again now.  I'm not sure if wiiuse had anything to do with it, but here's what happened in case it helps anyone.  
wiiuseexample requires that the board is not connected and that you press the red button to initiate connection.  On connection, my board was incorrectly identified by wiiuse as a wiimote, so I couldn't interact with that software.
I ended up pairing the board with an actual wii (within the wiifit app).  When I did, the wii reported that the batteries were too low! Doh! I switched batteries, then I measured myself on the wii to confirm it was working, then came back to xwiimote and it worked.
So, seemingly a schoolboy error on my part for not checking the batteries!  
However I'm fairly sure I did try changing them when the board first started playing up.   I actually suspect that it was the installation of some dependent package on the way to compiling wiiuse that made things work ... eg. libbluetooth-dev

blendmaster, 2018-05-30:
>I'm not sure if wiiuse had anything to do with it

Yeah, I think my problem was also hardware, maybe also the batteries being low -> no more balance board events. I don't actually have my actual wii set up though so I couldn't check by official means.

IIRC xwiishow's own battery indicator was nominal though, so perhaps there's a related problem where that doesn't work. I'll have to check once I get back to my balance board project.