xwiimote/xwiimote #88: New wiimotes incorrectly report zero position IR data initially

hifi, 2019-05-19:
While working on a project that uses the raw IR data from the Wii remote I noticed the initial values of the IR sensors are zeroed out (0x0x0) instead of set to off-screen values (1023x1023x0).

This can be tested with `xwiishow` when powering on the controller and looking at the data before any IR light has been seen. The values are one-by-one corrected from `0000x0000` to `N/A` when IR light comes and goes.

I had to add a special case of ignoring the zero position of all sensors as it was causing issues.

v1993, 2019-08-28:
This is an actual bug. I think that possible workaround for now it to check that values aren't zeros too (doesn't look like a big loss anyways).

R0NAM1, 2021-04-17:
> I had to add a special case of ignoring the zero position of all sensors as it was causing issues.
May I ask how this was done? I am having the same issue and it seems that this would be the only fix.

hifi, 2021-04-17:
There's nothing special to it, just ignore values that are zero in your code and assume it's offscreen.