xwiimote/xwiimote #89: How to use Nunchuk?

v1993, 2019-08-27:
I've been trying to get nunchuk extension working but when I move it's stick/press any button on it/shake it, no events seem to get fired.

I've tried opening all devices, but it have no effect. Also, I can't find anything in documentation what says how to handle nunchuk.

Is nunchuk supported? How can I use it?

v1993, 2019-09-15:
So it is implemented here in code, but is not documented (easy to figure out tho) nor included into Ubuntu release of this library. What a shame :angry:

moiri, 2021-04-09:
Are there any plans to release an updated deb package?

mrixner, 2024-12-24:
Hi, sorry to bump this super old post but have you gotten it working? I've been trying to figure this out for two days. My Wii remote (RVL-CNT-01-TR, if that matters) has only successfully connected to my (Manjaro Linux) computer via bluetooth a couple times, but every time it does, despite `xwiishow` reporting a nunchuk connected (`ext: nunchuk`)*, the nunchuk doesn't seem to fire any events. The Steam controller configurator** recognizes the main controller's inputs, but nothing seems to recognize the inputs of the joystick. I'm not sure if this was due to a mapping issue, but I couldn't find how to fix that. If it is a mapping issue, maybe the events _are_ being sent? Unfortuanately, I have no idea how to determine if that is the case. I have scoured the internet for any solution and nothing I've tried has worked. Maybe I haven't looked hard enough; can anyone point me in the right direction?

Note: I've seen that `cwiid` supposedly works, but I couldn't test it as `yay` wouldn't build it. I've also seen `wiican` that seemed promising, but it again relies on `cwiid`.

*[image]
**[image]

v1993, 2024-12-24:
Steam isn't going to have a good time with wiimotes, sadly - from its perspective, nunchuck is a separate input device.

mrixner, 2024-12-24:
That's unfortunate, but seems to resonate with other things I've encountered. So that means the nunchuk is active for anything built to take advantage of it correctly, but not Steam? 

Any other solutions you might've stumbled across? I have it working on Windows through a program translating it's calls into standard XInput calls. Is there anything similar for Linux or is this a lost cause?

v1993, 2024-12-24:
You may want to try something like evsieve that's capable of merging multiple evdev devices into one. Besides that, a specialized program is very well doable, but someone would have to write it first.

mrixner, 2024-12-24:
Soooooo.... nothing simple. Oh well! I guess I'll have to go do some research on getting a nunchuk's inputs recognized; I didn't realize it acted like a second device and I have no idea how to test / read it, but I'm sure Google has something for me. Thank you so much for your help!

marvin2k1, 2025-04-05:
There is both devices in one input event
https://github.com/gatopeich/Wiimote-2022

mrixner, 2025-04-06:
> There is both devices in one input event https://github.com/gatopeich/Wiimote-2022

Thank you very much for the suggestion, but it doesn't seem to work for me. I got to the `sudo insmod hid-wiimote.ko gamepad=1` step but it just tells me `insmod: ERROR: could not load module hid-wiimote.ko: No such file or directory` despite `dkms status` saying

```
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/Wiimote-2022/20.22/source/dkms.conf)
Wiimote-2022/20.22, 6.6.54-2-MANJARO, x86_64: installed (Original modules exist)
```

I then got the wiimote to connect with `bluetoothctl` but the remote is stuck flashing all four lights and Steam says no controllers are connected. If you had a quick fix, I would appreciate it!

marvin2k1, 2025-04-06:
Hello @mrixner 
I Answer you on apropiate git issue space:
https://github.com/gatopeich/Wiimote-2022/issues/3