xwiimote/xwiimote #90: Cannot build xwiimote

tim-blackmore, 2019-12-23:
I am on ubuntu 18.04.3 and I am struggling with installing xwiimote. My problem seems to echo similar problems in this thread but I have been unable to resolve the issue.

I am following the instructions and when I run the ./autogen.sh for xwiimote, it fails with the following lines.

./configure: line 13319: syntax error near unexpected token `UDEV,'

./configure: line 13319: `PKG_CHECK_MODULES(UDEV, libudev)'

I'd appreciate any advice on how to resolve this issue.

mira-tech, 2020-03-04:
I was so too.(I don't use ubuntu)
So I was able to install using apt-get! (I don't know why)
```
$ sudo apt-get update
$ sudo apt-get install  xwiimote
```

PureTryOut, 2021-12-28:
You're missing the development headers for udev, install them.