xwiimote/xwiimote #92: Nunchuck conflics with motion plus

cipitaua, 2021-11-03:
In my system (ubuntu devel) I get that nunchuck does work only without motion plus. When motion plus is connected (or integrated) the nunchuck is not detected at all.

Note that I have compiled xwiimote from github repo.

cipitaua, 2021-11-06:
no one?

git-developer, 2023-04-08:
I have two MotionPlus adapters. The first one is working fine, the second one behaves as described here.

I found 2 indications that my second device is not a genuine MotionPlus but a clone:
1. The screws have Phlilips (cross) heads; original MotionPlus screws have triangle heads.
2. The colour of the plastic inside the small connector that is plugged into the Wiimote is black; on the original MotionPlus it's white

`udevadm monitor -k` shows different behavior for the affected MotionPlus adapter:
* Plugging the MotionPlus adapter without Nunchuck into the Wiimote causes 2 `add` events (one for `inputX` and one for `inputX/eventY`; unplugging causes 2 `remove` events.
* The same happens for the Nunchuck alone without MotionPlus.
* When the MotionPlus adapter is connected and the Nunchuck is plugged into the Wiimote, `udevadm` shows
    * 2 `remove` events for the MotionPlus adapter
    * 2 `add` events for a MotionPlus adapter
    * 2 `add` events for a Nunchuck
    * 2 `remove` events for the Nunchuck
    * 2 `remove` events for the MotionPlus adapter

My guess is that the kernel module runs into an unexpected situation and thus removes the devices. To confirm this, a hid-wiimote module is required that is compiled with debug support.