xwiimote/xwiimote #94: Support Wiimotes connected via the DolphinBar

PureTryOut, 2021-12-29:
The [DolphinBar](https://www.mayflash.com/product/showproduct.php?id=6) is a Wiibar with USB-A connector. Besides doing the usual infra-red lights it also acts as a Bluetooth receiver the Wiimotes can connect too, which it then passes to the connected PC. The Dolphin Emulator has support for this built-in and works flawlessly, but it seems xwiimote doesn't support this.
My Wiimote is connected to it and although the Dolphin emulator indeed picks it up, xwiimote doesn't seem to realize a device is connected. Nothing appears in dmesg and `xwiishow list` doesn't list the device.

I'm willing to buy such a DolphinBar for the maintainer of this project if it helps getting it supported.

nadenislamarre, 2021-12-31:
my understanding of the dolphin bar is that it transform your wiimote as a mouse.
so, it is normal that xwiimote doesn't see it. instead a mouse i seen.
don't hesitate to buy one to me ;-)

PureTryOut, 2022-01-02:
That is a wrong understanding, it does much more than that.

It has 4 operating modes, the activated one being indicated by a LED:

- Keyboard and Mouse Mode (LED1)
- Keyboard and Mouse Game Mode(LED2)
- Game Controller Mode (LED3)
- Wii Remote Controller Emulator Mode (LED4).

The Dolphin emulator uses mode 4, and I honestly expected xwiimote to just work with it but it seems it doesn't. Looking into it, Dolphin had to make some changes to properly support it, so I suppose the same is the case for xwiimote.

A possibly relevant PR on the Dolphin side would be https://github.com/dolphin-emu/dolphin/pull/661.

Efemache, 2023-06-03:
Hi,

@PureTryOut did you find a library supporting wiimote via dolphinbar ?
I'm looking for it too.
Pygame can manage wiimote (including the last model with integrated motion plus) via DolphinBar using mode 3 but I need the mode 4 to manage the drum from Guitar Hero.

PureTryOut, 2023-06-03:
Sadly no