xwiimote/xwiimote #98: XWII_EVENT_KEY is send 4 times for every button press

PureTryOut, 2022-01-06:
However, this should of course only be sent twice: once for the button down and once for the button up events.

v1993, 2022-01-08:
What are "status" fields? If it as auto-repeat, this is intended. Additionally, this might be a bug in driver rather than library.

PureTryOut, 2022-01-08:
Sorry, I do not understand what you mean with "status fields"?

v1993, 2022-01-08:
Ah, sorry for being unclear, there's a "state" field for key events, which determines type of event (up/down/autorepeat)/

PureTryOut, 2022-01-08:
Ah those, yeah that's just twice down and twice up. Also in that order, "1 1 0 0".

v1993, 2022-01-08:
Interesting, I can't reproduce this issue myself - events fire only once no matter what I try.

Can you run `evtest` on main wiimote device (`/dev/input/event[number]`) and see if that has duped inputs as well? Also, are you using git version of xwiimote or one from system?

PureTryOut, 2022-01-10:
Using the latest git version, as the last release lacks support for quite a few extensions it seems. And well, in general it has been ages since any commit to this project (sadly) so why not try whatever the latest is :shrug: 

`evtest` interestingly enough only sends a single up and a single down event for each press. Not sure what is going on here...

PureTryOut, 2022-01-10:
I'm marking this as "user problem". I changed the way I poll for new events and now I only get 2 events for each button press, 1 down and 1 up, like it's supposed to. Not sure what was happening before...

closed 2022-01-10 by PureTryOut