xwiimote/xwiimote #101: New release

PureTryOut, 2022-08-16:
The last release (xwiimote 2) doesn't contain all changes in git master, but some applications (like https://invent.kde.org/plasma-bigscreen/plasma-remotecontrollers) depend on that functionality. It has been absolute ages since the last commit and release, but I'd love to see a new release which includes these changes so distros can update their packages.

dev-0x7C6, 2022-08-18:
@dvdhrm Hi, I would like help maintain project, I'm having some objectives that might not align with original authors intent, i.e. I would like to port project to cmake build system. If this is not acceptable I will go with fork xwiimote-ng or something like that.

dev-0x7C6, 2022-08-19:
I decided to fork as https://github.com/dev-0x7C6/xwiimote-ng, I will try to stay fairly source compatible as long as I can.

Current changes:

* provided initial configuration for cmake build system
* merged most stuff from pending reviews @PureTryOut  @sahoahfoa @v1993 as is
* formatted source and added configuration for clang-format that is fallowing xwiimote code style

My future goals are:

* provide C++ headers / wrappers / library ontop base xwiimote-ng (optional target)
* provide get function for wiimote rumble state
* improve build system, separate targets, make more optional targets (i.e. for tools)
* maintain release schedule

PureTryOut, 2022-08-19:
Since this seems fairly dead sadly, I'm glad someone is picking up the work, thanks!

It would be nice if after the change to CMake you could make an initial release so distros can start picking it up.