xwiimote/xwiimote #102: List of supported third party wiimotes?

cdfh, 2022-09-11:
Given that Nintendo have sadly discontinued the official wiimote and old stock is drying up, we (the Linux wiimote community) are increasingly having to make do with third party implementations. However, it can be difficult to find third party wiimotes that work with xwiimote. I've just bought two, both popular on Amazon, but sadly both failed to work (one failed to appear in bluetooth scans, the other appeared and could be paired, but didn't show up in `xwiishow list`). While third party devices are broadly [expected to be supported](https://dvdhrm.github.io/xwiimote/#devices), it has perhaps become understandably difficult to support them all.

With that in mind, I believe it would be very beneficial to maintain a list of third party wiimotes together with their known statuses with respect to xwiimote-compatibility. I would propose a wiki page would be best suited to this, but in lieu of a wiki page (as I don't think the current project has a wiki configured), I would propose that an open/on-going Github Issue would be more than adequate, at least to get the party started.

Sadly, as I only have two failed third party wiimotes, I can't kick-start the list with a working example, but I wonder if anyone else might be able to?

Lazorne, 2024-01-02:
There have been some efforts over at to get it working:

https://github.com/xwiimote/xwiimote/issues/66
https://github.com/bluez/bluez/issues/94

And things are still happening but they are moving slowly. 
But the best solution would be to have a community effort to first be recognized by Bluez and try to get as many types as possible. 

You already can use them for example with Dolphin on Linux by using the Bluetooth Pass-tough method.
https://wiki.dolphin-emu.org/index.php?title=Bluetooth_Passthrough

chetto1983, 2025-01-03:
I'm having the same issue with 3rd party wiiMote 2 in 1. Dolphin detect but bluez doesn't.  There ara something we can do to make it work?