xwiimote/xwiimote #103: buttons dont do anything?

qube-03, 2023-01-28:
hi, im using this to use my wiimote as a remote for kodi.

im on Raspberry Pi OS Lite 32 Bit

ive installed xwiimote and the Kodi add on controller profile for the wiimote.

the arrow buttons work just fine, however the A, 1 and 2, B and Home buttons do nothing. It doesnt even seem like theres any input being brought to the screen. xwiishow isnt showing me anything either. the a button is supposed to be mapped to enter according to what ive read in the manpage, yet still nothing happens. im at a loss

any help would be awesome, thanks!

nadenislamarre, 2023-01-29:
did you put the wiimote in joystick mode ?
by default it is in gun mode. what is the icon on top left ?

qube-03, 2023-01-29:
im not sure what top left you mean, heres a screenshot of xwiishow though, hope this helps!

nadenislamarre, 2023-01-29:
top left in emulationstaton. the pad icon. a gun or a pad ?

qube-03, 2023-01-29:
sorry, im not familiar with emulationstation, is that a program i can just install or does it refer to any emulator?

nadenislamarre, 2023-01-29:
see
https://wiki.batocera.org/_media/blog:screenshot-2021.08.28-15h15.03.jpg?w=720&tok=f2a739

es aka emulationstation is the main batocera program

qube-03, 2023-01-29:
ok so i compiled it and did all the stuff, just did `touch test.nes` to make a fake nes game so i can actually enter the program.

upon loading it wants me to configure my controller. i press on the wiimote and then input my buttons.

the wiimote though is being detected as a keyboard. A is enter, B is space and so on. im not quite sure what to do now, the xwiimote driver is making the wiimote be recognized as a keyboard.

also theres no icon to be seen, only a white background:
[image]

nadenislamarre, 2023-01-29:
ok, sorry. i think it was an other github project. this one is not related to es. so, forget all what is said.
i just received recently a third party wiimote (not official one) and it seems that recent 3rd party wiimote doesnt react the same as official wiimotes (but are working on wii however). I've not yet worked on that.
what's your wiimote ?
can you check if some events happen thanks to evtest ?
(forget emulationstation)

qube-03, 2023-01-29:
its a normal wiimote with wii motion plus inside, not 3rd party. ill try evtest

nadenislamarre, 2023-01-29:
with the nintendo logo ?

qube-03, 2023-01-29:
yea its on the back cover, an official remote.

i have figured out tho that this is likely to be a problem with kodi itself, as the wiimote works perfectly everywhere else

nadenislamarre, 2023-01-29:
ah ok.
then maybe i know your issue.
the wiimote is not a joystick nor a mouse on linux.
- can you confirm that you get events ?
- if yes, you must add udev rules to make it as a joystick or a mouse by adding ID_INPUT_JOYSTICK or ID_INPUT_MOUSE on the ir.

qube-03, 2023-01-29:
i dont even necessarily need the wiimote to be a joystick or mouse.

as said, kodi uses enter and such to go through menus. pressing A on the wiimote with the xwiimote driver DOES press enter, but kodi for some reason doesnt recognize it at all.

nadenislamarre, 2023-01-31:
kodi code, like most of sofware bases on the ID_INPUT_JOYSTICK ID_INPUT_KEY/KEYBOARD ... you just need udev rules.