xwiimote/xwiimote #117: Cannot pair Wiimote

WilsontheWolf, 2025-02-27:
When trying to pair with the wiimote, it immediately shuts off. Same behaviour with bluetoothctl and kde's bluetooth. The controller does work fine in dolphin-emu, though I can't use it for the system when it's paired to dolphin.

Fedora 31
Linux 6.12.11-200.fc41.x86_64

https://github.com/user-attachments/assets/0ae623c2-1fb5-4d29-b5ee-a66cd6979a0b

D-Skye, 2025-03-18:
I have the same problem, but I managed to get it to sync by pressing the sync button again when the lights turned off after selecting the wiimote from the list of devices (around the timing where you pressed the A button in your video).

OrdinaryCloverOC, 2025-03-27:
br-connection-create-socket error on endeavouros